nicinabox/lets-split-guide #2: Re: Diodes

Khord, 2017-03-19:
> It doesn't actually matter which side you mount the diodes on so long as they're in the correct orientation.

It's best practice to mount them on the bottom side so that if one goes bad after a while, you can desolder/replace it without having to desolder all your switches to remove your plate and access the top side.

nicinabox, 2017-03-20:
Makes sense. I'd like if the photos matched what's written in the text. We'd need to source new photos at that build stage to fully make this change.

Khord, 2017-03-20:
Ok, I actually am about to build one, just waiting on parts. So I can take some build photos for this and probably issue #3 too.

nicinabox, 2017-03-20:
Sweet. Just open a PR when you're ready.

benji877, 2017-05-24:
It is actually best to mount the diodes on the top if you are using 3mm acrylic, if you don't your plate will move up and down. The diodes keep it pushed against the bottom of the switches.

closed 2017-06-17 by nicinabox