nicinabox/lets-split-guide #3: Re: RGB Underglow

Khord, 2017-03-19:
https://camo.githubusercontent.com/6251f238a6d77ca203f50d6ee6f5ea8e0a81675b/687474703a2f2f692e696d6775722e636f6d2f6d5379704335792e6a7067

It is redundant to solder the 5V and GND to the pro micro. Only the data is directional; the power will flow from your solder points at the TRRS jack.

http://i.imgur.com/Uj0o7qr.jpg

nicinabox, 2017-03-20:
👍 I'd actually updated my wiring to this a while ago and have been meaning to take some new pictures.

Xyverz, 2017-03-20:
I have a question about this - can the power wires that you're connecting be on any part of the LED strip, or is it necessary to have them somewhere in the middle and not at the far end? It might be good to clarify this.

nicinabox, 2017-03-20:
@Xyverz VCC and GND can be anywhere on the strip. DO needs to be on the last one. I'm more a fan of putting all the connections together, but splitting them up is fairly common.

closed 2017-06-17 by nicinabox