nicinabox/lets-split-guide #4: link to source PCBs

jimmyhchan, 2017-04-27:
While i strongly suggest promoting switchtop and their service, unfortunately they are currently out of stock.

So it looks like the PCB source (KICAD) and gerber files have been made available so anyone can order their own from their favorite pcb shop.

https://github.com/climbalima/let-s-Split-v2

The Gerber zip file with instructions like http://www.40percent.club/2017/03/ordering-pcb.html (SHOULD -- assuming the gerber zip is correct) gets you about 10 (blue/black) boards (5 pairs) for $37 shipped USA ($7.4 each pair)

nicinabox, 2017-04-27:
👍 Want to submit a PR for this change?

jimmyhchan, 2017-04-27:
Will do! I wanted to get the issue up first in case this was already in the works.... then I got lost in kicad.

ipetepete, 2017-05-18:
EasyEda requires the dimensions of the PCB to order. Does anybody have this information? I cannot find it anywhere. I'm guessing the schematics have the dimensions, but I don't have the software required to open them.

jimmyhchan, 2017-05-18:
See my pull request for this. #5  Alternatively start the process of
ordering on oshpark and you'll get the size.

GlenPickle, 2017-05-19:
I uploaded the zip files to OHS park and they said the dimensions were 112.17x79.17mm. This seems really small to me though... so I might just wait for Switchtop to get them back in so I know I'm getting the right size.

ipetepete, 2017-05-19:
@GlenPickle I had them measured to be sure, those are the correct measurements. I'm not sure switchtop will be stocking them anymore, have you heard that they will?

GlenPickle, 2017-05-19:
@ipetepete Excellent, thank you for letting me know. I want to make more than one so I will go ahead and order them from EasyEDA.

As of May 18, 2017, Switchtop's "Restock Status" for the PCB's states that a new batch has been ordered, and they will be in late May or early June

closed 2017-05-25 by nicinabox