nicinabox/lets-split-guide #12: ProMicro Slave Jumper Clarification

zachfedor, 2017-06-22:
In the [assembly instructions](https://github.com/nicinabox/lets-split-guide/blob/master/assembly.md#one-side-isnt-working), you have a troubleshooting note about the bridged J1 jumper on the Pro Micros. I'm running into this issue, where the master side works fine but there is no signal from the slave.

I'm using SparkFun's Pro Micros and both have shorted J1s. Should I remove the solder bridge from one or both of the micros?

nicinabox, 2017-06-22:
I believe so. I haven't tested it on a Sparkfun version, but I did test it both ways on the common clone version and it wouldn't work when jumpered. The schematic shows a closed J1 bypasses a bunch of stuff and puts it into 5V mode. I don't really know why 5V doesn't work though.

Worst case if you find theres no change, just put a dab of solder back on it.

[image]

zachfedor, 2017-06-22:
good point. i'll give it a shot tonight and update with my findings

zachfedor, 2017-06-23:
No luck.

I removed the solder bridge from both Pro Micros and I'm still not receiving any input from the right hand side. Ongoing discussion is happening in [help a help thread I started on /r/olkb](https://www.reddit.com/r/olkb/comments/6hxawx/help_with_lets_split/). Someone else is having the same issue, too, and we're talking it out [here](https://www.reddit.com/r/olkb/comments/6ixnip/help_flashing_lets_split/).

nicinabox, 2017-08-02:
@zachfedor any updates on this?

zachfedor, 2017-08-02:
Kind of. When the left hand was master, I couldn't get any signal from the right hand. The ProMicros LEDs were constantly lit, which I believe is a good hint that something ain't right. When I plugged the right hand in as master, I could get key presses from the left hand as a slave. Obviously the default keymap was mirrored this way, but at least something was working.

I eventually gave up trying to find the issue. Instead, I modified my keymap to force right hand master and can only plug it in from that side now. I also had to change the function that determined which side had the USB plugged in, which seemed counter-intuitive. But it works!

Here is the commit containing those code changes: https://github.com/zachfedor/qmk_firmware/commit/42655a4870ad73ecbce4c663957c1cd3f4baa773

nicinabox, 2017-08-02:
Cool! I'm going to close this for now but hopefully this is helpful to someone.

closed 2017-08-02 by nicinabox