nicinabox/lets-split-guide #15: Connecting the jumpers

inucode, 2017-07-24:
The assembly instructions, under the "Connect Jumpers" section, say to "Do both PCBs the same", which is correct if you have one PCB flipped over, but if you connect the exact same jumpers on both PCBs it ends up not working (found that one out the hard way). I'm not sure exactly how to make this more clear but more detail or notes in the troubleshooting section might be helpful. I'd attach a photo of my other PCB showing the correct jumping but my soldering is terrible, haha...

That said, this was my first real "DIY" electronics project, and the guide was great and easy to follow, thanks for putting it together!

Xyverz, 2017-07-24:
Does this help clarify?

inucode, 2017-08-01:
I think part of the confusion for me is that the 'underside' of the PCBs are two different sides. Maybe specifying that it should look like that on the side that the TRRS connector is mounted on would help more?

Xyverz, 2017-08-02:
One would assume that the "underside" of the PCB is the side that doesn't have the switches. I'm not completely certain that it needs to be clarified much further. 

I believe, however, that we should have a photo of that part of the pcbs next to each other so that future users can know what to look for. I'm about to build another Let's Split, and I'll make sure to get a photo of this for the documentation.

nicinabox, 2017-08-02:
Clarified "underside" in https://github.com/nicinabox/lets-split-guide/commit/51ee26dcb23086e3aa0896bd5ee34c2051a7043f.

closed 2017-08-02 by nicinabox