nicinabox/lets-split-guide #18: Improve wording on step "Mount the 2 switches under the Pro Micro"

rlbaxter, 2017-09-07:
When I read "2 switches", I grabbed 2 switches and mounted one on each side. It was not clear to me that I needed to mount 4 switches, 2 per side until after I had mounted the pro micros and found I couldn't access the other spot on each side. I felt pretty stupid after. Could we add some clarification here so others don't make the same mistake? A picture or a diagram might be helpful too.

Xyverz, 2017-09-07:
> Sorry about the goof up. I'd be happy to add an image in for this but my
> LS is fully assembled so someone else would need to provide a picture at
> that stage.

I'll send a photo after I get home; I have another set of PCBs waiting to
be built out.

Duckle29, 2017-12-25:
@rlbaxter Does the wording in the current assembly guide help? if so @nicinabox can close this issue :)