nicinabox/lets-split-guide #32: two halves work alone, but not with cable

adkron, 2018-03-30:
The two halves of the keyboard work when plugged directly into the computer, but once I add the trrs cable everything stops working. I didn't know where else to go for help, and I thought it might need to be added to the troubleshooting doc if we can figure it out.

nicinabox, 2018-04-01:
@adkron Can you show a picture of your TRRS cable?

adkron, 2018-04-02:
I added the jumper picture too because I’m thinking it might be wrong. When the guide says to do them the same I thought of stacking them and making sure they were identical. Now I’m thinking it should have been the opposite.

adkron, 2018-04-02:
Okay, I switched the jumpers and everything works. I'm not sure how to make the docs more clear other than to have a picture of the two halves side by side.

Now I need to figure out how to remove the pro micro. My left-hand side has one switch that isn't working, and it is the top one above the pro micro. 😢 I checked all the switches with a multimeter before starting. I probably need to reflow it, but I'm not good at removing entire PCBs. Wish me luck.

nicinabox, 2018-04-02:
> Okay, I switched the jumpers and everything works.

Excellent. I think the docs say do them the same, but for some reason that has tripped up some people. They should be identical on both sides.

> My left-hand side has one switch that isn't working, and it is the top one above the pro micro.

Sometimes those switch pins can short against the surface of the pro micro. Try bending them away with a small tool. If that doesn't change anything you may have to desolder :(

Good luck!

closed 2018-04-02 by nicinabox

adkron, 2018-04-02:
Thanks for the help. I will take a picture of the correct soldering for the jumpers and make a or to add them to the guide. That will clear things up. Nothing like a picture. 

Amos King
Binary Noggin