nicinabox/lets-split-guide #38: dongle

dracolyteking, 2019-05-09:
Sorry not really an issue but i was wondering what the little dongle plugged into the micro in the photo for the buzzer was for.

nicinabox, 2019-05-09:
I think you're referring to [this page](https://github.com/nicinabox/lets-split-guide/blob/master/speaker.md)? @rickard-von-essen contributed that, perhaps he would know?

dracolyteking, 2019-05-09:
Yeah, the little silver micro usb dongle in the micro.

rickard-von-essen, 2019-05-09:
Hi, the micro USB connector on Pro micros (especially clones) are renowned for breaking off after some use. So to minimize wear I got a _detachable USB cable_, something similar to [this](https://m.aliexpress.com/item/32858924703.html?sellerAdminSeq=229251348&spm=a2g0n.search-amp.list.32858924703&aff_trace_key=&aff_platform=msite&m_page_id=1030amp-FHCOQj0ieGapDUVehMWo2Q1557438733058).

dracolyteking, 2019-05-09:
Ah thanks

closed 2019-05-14 by nicinabox