nicinabox/lets-split-guide #39: Updated plate dimensions

grahampfau, 2020-06-30:
In the assembly instructions, ordering a case mentions some dimensions:
> Select Choose a size > P1 - 7.126 inches long x 7.126 inches wide

However, when looking at the .svg, it's not square, but rather the top and
bottom plates are in the same .svg, making it overall taller than it is wide.

What are the correct dimensions to specify when ordering the plates cut?

rickard-von-essen, 2020-06-30:
Each plate should be something like 12 cm x 8.5 cm.
[image]
[image]

grahampfau, 2020-06-30:
> Each plate should be something like 12 cm x 8.5 cm.

I actually already have a set of (broken) acrylic plates that measure 116mm wide,
but that's not super exact, because my measurements seem to drift by 0.5mm or so
each time I measure. I figured I'd get as close to a "true" number as possible, since
the dimensions need to pretty exact for switches to fit correctly.

rickard-von-essen, 2020-06-30:
The only really important measurement is that the holes for the switches should be 14x14 ±0.05 mm.

See https://matt3o.com/anatomy-of-a-keyboard/

grahampfau, 2020-06-30:
[image]

I'm not sure how to use vector editing programs to determine what I need to set the overall width here to based off the measurement of 14mm key switch cutouts.

grahampfau, 2020-07-14:
Just got the plates I ordered with the above dimensions and everything fits.

rickard-von-essen, 2020-07-14:
> Edit: Just put in an order with the dimensions of `116.3mm x 174.1mm x 1.5mm` and it's set to arrive in late July. I'll update this issue with the results and a PR if it works out.

Shouldn't it be 3 mm thick?

grahampfau, 2020-07-14:
For acrylic, yes. I went with metal, so 1.5 mm was perfect. Since the guide mentioned acrylic, my PR has 3 mm.

hijackeel, 2021-04-28:
I just received 116.3mm x 174.1mm x 3.2mm maple plates from Ponoko and can confirm that everything fits together perfectly.

grahampfau, 2021-05-25:
@nicinabox do you think we can get this closed out and have https://github.com/nicinabox/lets-split-guide/pull/40 mergedin ?